Floyd-Fish/ST7789-STM32 #3: about font generated question

liukai-tech, 2019-08-15:
Hello,
   I can run normally with your driver, but now I need to change the ASCII font size, Chinese character set and modify the picture. I would like to ask what software you use for module extraction and generation, could you tell me the specific software and procedures of module extraction? Thank you very much!

Floyd-Fish, 2019-11-19:
中文字体和ASCII用最常见的那种取模软件就可以，图片我用了一个py脚本取模（RGB565）

Lucy2003, 2020-08-01:
您好，请问如何实现输出特定背景颜色和前景颜色的中文字符（串）和 
自定义宽高(16x8)的英文字符串（库里没有这个，只有7x10）？
（为了节省空间，还是不把字符取成图片比较好。）

另外，如何利用 ST7789_DrawPixel_4px() 来放大字体？

谢谢。

Floyd-Fish, 2020-08-04:
1.请看显示字符/字符串函数的原型，是有`bgcolor`(背景颜色)参数的，如果想要显示中文，是需要自己取模的呢。。除非外挂一个中文字库

2.我想到的在MCU平台上显示字符的方式只有对字符bitmap取模。。矢量字体反而更为复杂了，所以如果你想用别的字体的话可以自己取模。。。

3.字符的显示并没有调用`Draw_Pixel`函数，而是直接将bitmap数组映射到显存中，所以用这一个版本的WriteChar函数是不支持缩放的，如果你想试着缩放字体，可以用DrawPixel来实现WriteChar功能，再用4px的函数来实现放大字体

Lucy2003, 2020-08-04:
用Draw_Pixel的话，效率也许有点低了。不过，我会试试的。谢谢。

Floyd-Fish, 2020-08-04:
我找到一篇介绍字体缩放算法的文章，使用了`nearest-neighbor resampling`，不过我没测试过，你可以看看
[链接here](https://www.davidepesce.com/2018/10/25/diyconsole-part-6-text-and-fonts/)

dimky, 2020-08-30:
Hi Floyd-Fish.
Could you please share python script to convert BMP picture into C array? Thanks!

Floyd-Fish, 2020-08-31:
Please refer to this repo:
[afiskon/st7735](https://github.com/afiskon/stm32-st7735/tree/master/img)

liukai-tech, 2021-01-08:
> 中文字体和ASCII用最常见的那种取模软件就可以，图片我用了一个py脚本取模（RGB565）

你好，请问能否告知中文字体及ASCII取模软件，使用PCtoLCD2002完美版无法正常取模，谢谢。

Floyd-Fish, 2021-01-08:
> > 中文字体和ASCII用最常见的那种取模软件就可以，图片我用了一个py脚本取模（RGB565）
> 
> 你好，请问能否告知中文字体及ASCII取模软件，使用PCtoLCD2002完美版无法正常取模，谢谢。

PCtoLCD2002我测试过可以正常取模，请检查你的取模设置是否正确，比如文件格式，取模走向，取模方式等

liukai-tech, 2021-01-08:
> > > 中文字体和ASCII用最常见的那种取模软件就可以，图片我用了一个py脚本取模（RGB565）
> > 
> > 
> > 你好，请问能否告知中文字体及ASCII取模软件，使用PCtoLCD2002完美版无法正常取模，谢谢。
> 
> PCtoLCD2002我测试过可以正常取模，请检查你的取模设置是否正确，比如文件格式，取模走向，取模方式等

我是采用 阴码+行列式+顺向+C51格式进行取模，但是生成的数据是uint8_t型，我看fonts.c内是uint16_t类型，这个软件还需要修改哪里？另外您取模用的是什么字体？不是太懂，谢谢您的解答。

Floyd-Fish, 2021-01-08:
> > > > 中文字体和ASCII用最常见的那种取模软件就可以，图片我用了一个py脚本取模（RGB565）
> > > 
> > > 
> > > 你好，请问能否告知中文字体及ASCII取模软件，使用PCtoLCD2002完美版无法正常取模，谢谢。
> > 
> > 
> > PCtoLCD2002我测试过可以正常取模，请检查你的取模设置是否正确，比如文件格式，取模走向，取模方式等
> 
> 我是采用 阴码+行列式+顺向+C51格式进行取模，但是生成的数据是uint8_t型，我看fonts.c内是uint16_t类型，这个软件还需要修改哪里？另外您取模用的是什么字体？不是太懂，谢谢您的解答。

贴一段示例，我还没测试过能不能用
```c
//取模汉字,添加至fonts.c/fonts.h中,16X16
uint8_t ST7789_16[]={
0x80,0x00,0x80,0x00,0x80,0x00,0x90,0x04,0x90,0x08,0x88,0x10,0x88,0x20,0x84,0x28,
0x82,0x08,0x80,0x04,0x00,0x02,0x00,0x01,0x80,0x00,0x40,0x00,0x30,0x00,0x0E,0x00,/*"少",0*/
0x40,0x00,0x40,0x00,0x40,0x00,0x40,0x00,0x20,0x00,0xFF,0x7F,0x20,0x04,0x10,0x04,
0x10,0x04,0x08,0x02,0x18,0x02,0x60,0x01,0x80,0x01,0x60,0x06,0x18,0x08,0x06,0x10,/*"女",1*/
0x04,0x10,0x08,0x38,0x88,0x07,0x9F,0x00,0x90,0x00,0x88,0x00,0x88,0x7F,0x9C,0x08,
0xAA,0x08,0xA9,0x08,0x88,0x08,0x88,0x08,0x88,0x08,0x48,0x08,0x48,0x08,0x28,0x08,/*"祈",2*/
0x04,0x04,0x08,0x04,0xC8,0x7F,0x1F,0x04,0x90,0x3F,0x08,0x02,0xC8,0x7F,0x1C,0x12,
0x2A,0x11,0x29,0x7F,0x88,0x10,0x88,0x12,0x48,0x14,0x28,0x10,0x08,0x14,0x08,0x08,/*"祷",3*/
0x80,0x00,0x80,0x00,0x80,0x00,0x80,0x00,0xFC,0x1F,0x84,0x10,0x84,0x10,0x84,0x10,
0x84,0x10,0x84,0x10,0xFC,0x1F,0x84,0x10,0x80,0x00,0x80,0x00,0x80,0x00,0x80,0x00,/*"中",4*/
};
```
然后是写汉字的函数,提供2种大小,分别是16x16和32x32,看你怎么取模了~

```c
//显示汉字
//index ->汉字的序号，需要在字库查找
//size ->字号, 16/32
void ST7789_ShowChinese(uint16_t x, uint16_t y, uint8_t index, uint8_t size, uint16_t color, uint16_t bgcolor)	
{  
	
	uint8_t i,j;
	uint8_t *temp,size1;
	if (size == 16)
		temp = ST7789_16;//选择字号
	
	if (size == 32)
		temp = ST7789_32;
	
 	ST7789_SetAddressWindow(x,y,x+size-1,y+size-1); 	//设置一个汉字的区域
  	size1=size*size/8;								//一个汉字所占的字节
	temp+=index*size1;								//写入的起始位置
	ST7789_Select();
	for(j=0;j<size1;j++)
	{
		for(i=0;i<8;i++)
		{
		 	if((*temp&(1<<i))!=0)	//从数据的低位开始读
			{
				uint8_t data[] = {color >> 8, color & 0xFF};
				ST7789_WriteData(data, sizeof(data));	//点亮
			}
			else
			{
				uint8_t data[] = { bgcolor >> 8, bgcolor & 0xFF};
				ST7789_WriteData(data, sizeof(data));	//填充背景色
			}
		}
		temp++;
	 }
	ST7789_UnSelect();
}
```

liukai-tech, 2021-01-16:
> 贴一段示例，我还没测试过能不能用
>
> ```c
> //取模汉字,添加至fonts.c/fonts.h中,16X16
> uint8_t ST7789_16[]={
> 0x80,0x00,0x80,0x00,0x80,0x00,0x90,0x04,0x90,0x08,0x88,0x10,0x88,0x20,0x84,0x28, [...]

closed 2021-01-16 by liukai-tech