Floyd-Fish/ST7789-STM32 #6: Frequency Problem

turgayhopal, 2020-04-16:
Hi,

I have tested your code but it only works with 4MHz. Is there any place you can help?

Floyd-Fish, 2020-04-19:
Maybe you should connect 2 pull-up resistor on SCLK and MOSI(SDA)...I have tested it on 42MHZ(stm32f4) with pull-up resistors and it worked fine.

turgayhopal, 2020-04-21:
Thank you very much for your interest. I will test and return soon.

turgayhopal, 2020-04-22:
Again Hi,

I tested what you said. I tried tensile resistors by connecting them to 5V. The values ​​of my resistors are 3.3K but did not work. It only works at 2MHz. Could there be a different place that I did wrong?

Thank you very much for your interest.

Floyd-Fish, 2020-04-23:
Pull-up resistors should be connected to 3.3v because stm32's IO is 3.3v logic level...And you may check stm32's clock tree configuration, I'll upload a demo project based on stm32f407vet6 later

turgayhopal, 2020-04-28:
Thank you very much for your interest, I have achieved a successful result thanks to you.

closed 2020-04-28 by turgayhopal

MrHertzz, 2022-03-10:
Hi,

I got a same problem. I tried to initialise on Stm32cubeIDE there is no error but also it isnt work :D

RSTurgay, 2022-03-12:
Melih Merhaba, 

Ben bu yazılım üzerinde bir kaç revizyon yapmıştım sonradan çalışmıştı verimli bir şekilde. Ayrıca hangi denetleyiciyi kullandın da çalışmadı ?

turgayhopal, 2022-03-12:
> Hi,
> 
> I got a same problem. I tried to initialise on Stm32cubeIDE there is no error but also it isnt work :D

Melih Merhaba,

Hangi denetleyiciyi kullandın ?  Ve bende bunun yazılım üzerinde bir kaç revizyon yaparak düzelttiğim versiyonu var bakmak istersen gönderebilirim.

MrHertzz, 2022-03-12:
Merhaba Turgay çok güzel olur gerçekten stm32f405 ve stm32f407 kullanıyorum
ve slave pinini kendim yapılandırmak istiyorum

Turgay HOPAL ***@***.***>, 12 Mar 2022 Cmt, 10:33 tarihinde
şunu yazdı:

MrHertzz, 2022-03-12:
Merhaba Turgay,

çok güzel olur gerçekten... stm32f405 ve stm32f407 kullanıyorum ve slave pinini kendim yapılandırmak istiyorum.

turgayhopal, 2022-03-12:
Merhaba Melih,

Buradan ulaşabilirisin.

https://github.com/turgayhopal/STM32-ST7789-Library/blob/main/README.md