Floyd-Fish/ST7789-STM32 #8: About STM32F103C8T6

etkanburoglu, 2020-06-24:
Hello,
I am trying to show pictures and videos on an st7789 screen with stm32f103c8. I wanted to try your code but my Keil uVision version does not support (L6047U: The size of this image (49960 bytes) exceeds the maximum allowed for this version of the linker) such long codes or such big codes. Can you give me any advice or hint? I need to show pictures as one function and show videos as a second one.
Best regards

turgayhopal, 2020-07-02:
Hello there,

I'd like to help.

I have developed several projects using this library, of course by organizing a few things by myself. I haven't made a video using the library before, but if you can send it frame by frame, you can view the video.

So what's your problem?

etkanburoglu, 2020-07-02:
Of course, you can show the videos by taking random 8-10 frames from the current 25-30fps ones. The problem is changing the dimensions of the videos and formats. Let say we have a .avi or .mp4 video and they are 800*600 1024*768 compressing them to 240*240 and showing them on st7789 is quite complex.  Also, the speed of the stm32f103 is quite slow.  I guess I will change it to black pill or esp32

modi12jin, 2020-08-30:
> 当然，您可以从当前的25-30fps帧中随机抽取8-10帧来显示视频。问题在于更改视频的尺寸和格式。假设我们有一个.avi或.mp4视频，它们是800 _600 1024_ 768，将它们压缩为240 * 240，并在st7789上显示它们非常复杂。另外，stm32f103的速度也很慢。我想我会把它改成黑色药丸或esp32

https://homemadegarbage.com/makerfabs05