Floyd-Fish/ST7789-STM32 #14: Additional Fonts

Jo-Dan, 2021-07-21:
I'm using a modified version of this library and I'm just wondering where you get the fonts in this format? I'm hoping to find one a bit larger but when I tried to implement modified fonts from other libraries all I seem to get is nonsense on screen.

vinothkannanagva, 2021-09-30:
Jo Dan bro your make it on your own ,

Jo-Dan, 2021-09-30:
Well there are all manner of tools and resources for 8 bit fonts so I figured it was worth asking if they had used something similar for the ones in this project. 
I very much doubt the majority of fonts of this kind are made completely by hand as that's simply impractical.

vinothkannanagva, 2021-09-30:
> Well there are all manner of tools and resources for 8 bit fonts so I figured it was worth asking if they had used something similar for the ones in this project. I very much doubt the majority of fonts of this kind are made completely by hand as that's simply impractical.

well said  i made some font on my hand if you want i will give you
only numbers i made

Jo-Dan, 2021-09-30:
Well I don't really need it anymore but I'm sure if you were to upload them someone might find it useful down the line.

fadi0albaghdadi, 2022-03-02:
HOW I CAN CREATE FONTS ALONE WHAT IS THIS PROGRAMS