Floyd-Fish/ST7789-STM32 #32: Please check CPAH CPOL  level --请检查CPAH和CPOL极性

Yushi-Xing, 2024-07-28:
经过我的测试，一些屏幕支持README中配置的CPAH和CPOL极性（1Edge，High）
其中分两种情况：
有一部分屏幕支持CPAH和CPOL极性（1Edge，High）与（2Edge，High），
有一部仅支持（2Edge，High）。
目前测试了四五款ST7789 240*240屏幕，其差异性与制造商有关，暂不知道具体原因。
[image]
[image]

另外 在STM32F103系列中，SPI1和SPI2在HAL库的使用也表现出差异性
         在测试时使用（1Edge，High）配置SPI1和SPI2，在部分屏幕测试上仅SPI2可以点亮屏幕，怀疑HAL库在配置时，对CPAH和CPOL处理稍有问题。尽管在逻辑分析仪上看起来两个SPI外设生成的波形近乎一模一样，但是在部分屏幕测试上仅SPI2可以点亮屏幕。
我使用的版本为CubeMX6.8.0
[image]
SPI1
[image]
SPI2

因此若无法点亮，推荐尝试
[image]
希望有人能解答一下我的疑惑

Ximenzhengmo, 2024-10-13:
thanks bro, your discovery help me a lot !!!!

igorpie, 2024-11-15:
Thanks a lot!
"2 Edge" set help me to start some displays from Ali

hongkou-chaozi, 2024-11-23:
Thank you for resolving my ques

Wang8477, 2025-03-08:
Thank you for solving my problem

KNgithub26, 2025-08-05:
Thank you for solving my ques,i just confusing why my display  keep blank.

wuhensoft, 2025-08-08:
对，我也遇到了这个问题，调整spi CPOL CPHA 可以解决问题。