Floyd-Fish/ST7789-STM32 #35: Doesn't work on stm32f411ceu6(black pill), anyone have working example for this board on CubeMX?

Yura1993, 2024-10-12:
Try to use this library and unfortunately it doesn't works:
[image]
[image]
Maybe someone try to use it and have working example?

Tbthanh, 2024-11-04:
I have it work in CubeIDE, you might want to set all those DC RESET CS pin to opendrain and use an pull-up resistor for those (I use 200ohm still okay). If you need I can give you further info?

Yura1993, 2024-11-06:
> I have it work in CubeIDE, you might want to set all those DC RESET CS pin to opendrain and use an pull-up resistor for those (I use 200ohm still okay). If you need I can give you further info?

Can you send to me working example? I mean cubemx project that builded for stm32f411

classAndrew, 2024-12-17:
I got it to work on the exact same board after some trial and error on a 240x320 TFT display.

The main issue seemed to be confusing instructions in the README to set clock polarity (CPOL) to _High_. I'm not sure why this is required but setting **CPOL to _Low_** seems to have fixed it. I'm not sure why this is necessary since other projects keep the clock polarity low. Maybe someone can elaborate on this.

Next try with DMA turned off. There's a macro definition that you can comment out in st7789.h. This was the first step for me to get a working display. 

To turn DMA back on, make sure you enable SPI modes to `Transmit Only Master` and add an SPI_TX DMA request. 

Also, keep lowering the SPI baud rate until you see the test on the display. For me, I had to lower it down to 3 MHz.

BTW, if none of this works, you can try [this fork](https://github.com/deividAlfa/ST7789-STM32-uGUI) of the project which includes the .ioc file and project code. I got this one to work as well.