Floyd-Fish/ST7789-STM32 #38: False setup of spi bus and delay after reset is not long enough.

SonnyAndersen, 2025-01-03:
I have tested and it was very unstable.

Other who have worked with the ST7789 (ATM0130B3) has clock polarity low.

[image]

Also DMA needs to be setup.

[image]

Better but still unreliable. using reset button could sometimes get it working correctly.

Last issue was the reset timing, or more precisely the delay after reset for ST7789 to come correct out of reset.

void ST7789_Init(void)
{
	#ifdef USE_DMA
		memset(disp_buf, 0, sizeof(disp_buf));
	#endif
	**HAL_Delay(20);**
    ST7789_RST_Clr();
    **HAL_Delay(20);**
    ST7789_RST_Set();
    **HAL_Delay(500);**

With does things is fixed then the driver works well.

(Forks has the same problem)

BR Sonny

W-Linus, 2025-01-05:
**Thanks a lot!** When I using this library to drive 1.47inch Screen on STM32G431CB, I've found the display very unstable. As you said, I tried to add more delay after reset, the screen is stabled now, I think the author should add more delay into the code.

petit-miner, 2025-02-05:
Clock Polarity needs to be High and Clock Phase needs to be 2 Edges.

KasaiYuno, 2025-05-18:
> Clock Polarity 需要为 High 和 Clock Phase 需要为 2 个 Edges。

You are a miracle doctor! I did what you said, and the screen immediately became stable.